igrr/esptool-ck #1: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed

CHERTS, 2015-01-26:
After the last change (d7dcbf0) utility esptool.exe ceased to flash flash, constantly this error occurs:

C:\Espressif> esptool.exe -cp COM2 -cd ck -ca 0x00000 -cf eagle.flash.bin -vvv

esptool v0.0.3a - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting port from COM1 to COM2
	setting board to ck
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat eagle.flash.bin success
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
		flushed 0 bytes
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	read 0, requested 1
		espcomm cmd: didn't receive command response
trying to connect
	flush start
	setting serial port timeouts to 1 ms
		flushed 0 bytes
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	espcomm cmd: receiving 2 bytes of data
		espcomm_cmd: received 2 bytes: 
		0x00 
		0x00 
		espcomm cmd: response 0x20120707
	flush start
	setting serial port timeouts to 1 ms
		flushed 0 bytes
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_open
Uploading 28960 bytes from eagle.flash.bin to flash at 0x00000000
	erasing flash
	calculated erase delay: 300ms
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	setting serial port timeouts to 300 ms
	read 0, requested 1
	setting serial port timeouts to 1000 ms
warning: espcomm cmd: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader
	flush start
	setting serial port timeouts to 1 ms
		flushed 0 bytes
	setting serial port timeouts to 1000 ms
	flush complete

igrr, 2015-01-26:
Whoops sorry, looks like I was testing an old version before commiting. Will fix.

igrr, 2015-02-04:
Based on some feedback from windows users, this seems to be fixed now. @CHERTS can you verify?
Here's a win32 binary: http://download.igrr.me/esptool.exe.

CHERTS, 2015-02-04:
Я собрал последнюю версию у себя и вчера провел тесты, в принципе все нормально, но ошибка espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed все равно иногда возникает.
Как правило она возникает при прошивке подряд нескольких файлов по разным адресам.
Например:
esptool.exe -cp COM2 -cd ck -ca 0x00000 -cf eagle.flash.bin -vvv
esptool.exe -cp COM2 -cd ck -ca 0x40000 -cf eagle.irom0text.bin -vvv
esptool.exe -cp COM2 -cd ck -ca 0x7c000 -cf esp_init_data_default.bin -vvv
esptool.exe -cp COM2 -cd ck -ca 0x7e000 -cf blank.bin -vvv

На этапе прошивки esp_init_data_default.bin или blank.bin возникает espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed

probonopd, 2015-02-09:
I also get this now on linux64:

```
esptool v0.0.3a - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to none
	setting baudrate from 115200 to 115200
	setting port from /dev/ttyUSB0 to /dev/ttyUSB1
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /tmp/build2675618669274960120.tmp/ASCIITable.cpp_00000.bin success
opening port /dev/ttyUSB1 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
trying to connect
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	espcomm cmd: receiving 2 bytes of data
	espcomm_open
Uploading 30752 bytes from /tmp/build2675618669274960120.tmp/ASCIITable.cpp_00000.bin to flash at 0x00000000
	erasing flash
	calculated erase delay: 600ms
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	setting timeout 600
	done
	setting timeout 1
	done
	espcomm cmd: receiving 2 bytes of data
	writing flash
...............................
	setting address from 0x00000000 to 0x00040000
	espcomm_upload_file
	stat /tmp/build2675618669274960120.tmp/ASCIITable.cpp_40000.bin success
	espcomm_open
Uploading 148472 bytes from /tmp/build2675618669274960120.tmp/ASCIITable.cpp_40000.bin to flash at 0x00040000
	erasing flash
	calculated erase delay: 2050ms
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	setting timeout 2050
	done
	setting timeout 1
	done
warning: espcomm cmd: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader
```

Note that before trying to upload, I had uploaded a sketch that apparently crashes, since its only output was

```
Fatal exception (0):
epc1=0x4024a8c0, epc2=0x000000, epc3=0x000000, excvaddr=0x000000, depc=0x000000
```
a couple of times.

JeroenBeemster, 2015-03-02:
Hi I am trying to use this tool. Seems exactly what I need to upload from my windows machine.
I got this message:
warning: espcomm cmd: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
Another uploader works fine. But got a GUI. 
Could you advise me, If you have little more details about the -cd parameter. [I do not understand which pins should connect what]

igrr, 2015-03-03:
@JeroenBeemster could you please try the latest build at:
[https://ci.appveyor.com/project/igrr/esptool-ck/build/artifacts
this time with -vv argument, and if it doesn't work, post the full output here?

JeroenBeemster, 2015-03-03:
Thanks igrr. The latest build worked fine. 
Could you take a look at the -q option. I think it is a little bit too quiet. (with this option, he doesn't do the job anymore). I will not use this anyway. So only to inform you.

Leaves me with the last question:
How to use the -cd parameter. 
how to connect wires with ck or wifio 
One of both seems a good solution to start uploading without manually reseting.

JeroenBeemster, 2015-03-28:
Hi Ivan,

You did a great job by creating arduino IDE and the core.
You have noticed me last week when I published my version on www.arduinesp.com.
I have to atmid yours is much more sophisticated.
I would love to continue with your core, and focus on the user side of arduino.
Would love to hear from you,

Regards,

Jeroen

From: Ivan Grokhotkov 
Sent: Tuesday, March 3, 2015 7:31 AM
To: igrr/esptool-ck 
Cc: JeroenBeemster 
Subject: Re: [esptool-ck] espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed (#1)

@JeroenBeemster could you please try the latest build at:
[https://ci.appveyor.com/project/igrr/esptool-ck/build/artifacts
this time with -vv argument, and if it doesn't work, post the full output here?

tripzero, 2015-06-04:
Having this issue with the latest git master trying to flash the sparkfun thing esp8266:

/home/kev/.arduino15/packages/SparkFun/tools/esptool/0.4.4/esptool -eo /tmp/build8853751530736463083.tmp/espstar.cpp.elf -bo /tmp/build8853751530736463083.tmp/espstar.cpp_00000.bin -bm qio -bf 40 -bz 4M -bs .text -bs .data -bs .rodata -bc -ec -eo /tmp/build8853751530736463083.tmp/espstar.cpp.elf -es .irom0.text /tmp/build8853751530736463083.tmp/espstar.cpp_10000.bin -ec 

Sketch uses 198,408 bytes (37%) of program storage space. Maximum is 524,288 bytes.
Uploading 31552 bytes from /tmp/build8853751530736463083.tmp/espstar.cpp_00000.bin to flash at 0x00000000
...............................
Uploading 166904 bytes from /tmp/build8853751530736463083.tmp/espstar.cpp_10000.bin to flash at 0x00010000
.....................................warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: didn't receive command response

More verbose:

esptool -vv -cd ck -cb 115200 -cp /dev/ttyUSB1 -ca 0x00000 -cf /tmp/build8853751530736463083.tmp/espstar.cpp_00000.bin -ca 0x10000 -cf /tmp/build8853751530736463083.tmp/espstar.cpp_10000.bin 
esptool v0.4.5 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from /dev/ttyUSB0 to /dev/ttyUSB1
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /tmp/build8853751530736463083.tmp/espstar.cpp_00000.bin success
opening port /dev/ttyUSB1 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_open
Uploading 31552 bytes from /tmp/build8853751530736463083.tmp/espstar.cpp_00000.bin to flash at 0x00000000
	erasing flash
	size: 007b40 address: 000000
	first_sector_index: 0
	total_sector_count: 8
	head_sector_count: 8
	adjusted_sector_count: 4
	adjusted_size: 004000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 10000
	setting character timeout 100
	done
	setting timeout 1
	setting character timeout 1
	done
	espcomm_send_command: receiving 2 bytes of data
	writing flash
...............................
	setting address from 0x00000000 to 0x00010000
	espcomm_upload_file
	stat /tmp/build8853751530736463083.tmp/espstar.cpp_10000.bin success
	espcomm_open
Uploading 166904 bytes from /tmp/build8853751530736463083.tmp/espstar.cpp_10000.bin to flash at 0x00010000
	erasing flash
	size: 028bf8 address: 010000
	first_sector_index: 16
	total_sector_count: 41
	head_sector_count: 16
	adjusted_sector_count: 25
	adjusted_size: 019000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 10000
	setting character timeout 100
	done
	setting timeout 1
	setting character timeout 1
	done
	espcomm_send_command: receiving 2 bytes of data
	writing flash
.................................warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: didn't receive command response
closing bootloader

I was getting the same thing when trying 0.4.4, so I downloaded the latest from git and shoved it in the 0.4.4 directory.  Any hints?