igrr/esptool-ck #2: Compilation fails with ‘CRTSCTS’ undeclared

probonopd, 2015-02-07:
On Ubuntu 14.04.1 LTS compilation fails:

```
me@host:~/esptool-ck$ make
gcc -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -std=c99 -Os -Wall  -DLINUX -c serialport/serialport.c -o build/serialport/serialport.o
serialport/serialport.c: In function ‘serialport_open’:
serialport/serialport.c:374:22: error: ‘CRTSCTS’ undeclared (first use in this function)
     term.c_cflag &= ~CRTSCTS;
                      ^
serialport/serialport.c:374:22: note: each undeclared identifier is reported only once for each function it appears in
make: *** [build/serialport/serialport.o] Error 1
```

probonopd, 2015-02-08:
This can be fixed by using the `-std=gnu99` compiler flag instead of `-std=c99` (but since I don't know what this does to other platforms, it is probably best to change the Makefile so that this flag is used only on Linux platforms).

igrr, 2015-02-10:
I've switched to -std=gnu99 for all platforms and it seems to be okay.

closed 2015-02-10 by igrr