igrr/esptool-ck #4: add auto-reset feature

nerdralph, 2015-04-06:
I've added support for auto-reset using an RC filter between Rx and RST, similar to what I did for the Arduino.
http://nerdralph.blogspot.ca/2014/02/zero-wire-serial-auto-reset-for-arduino.html
I tested it with a 7.5k resistor between Rx and RST, a 4.7uF cap between RST and VCC (no diode required).  It's a bit oversized; anything with a RC constant of 10ms or more should be fine.  To be sure a long series of zero bytes won't trigger it, a diode in series with the resistor can be added.

Built with gcc 4.8.1/MingW and tested under Win7E/64, and it works great.  It has the added bonus of improving stability - I don't get espcomm_sync failed any more.  This is likely because the cap on the reset line filters noise that can cause spurious resets.

commit e2a0321446d301cdf4afb1c91327211e25afa1d8
Author: unknown <ralphdoncaster@gmail.com>
Date:   Sun Apr 5 22:51:30 2015 -0300

    add auto-reset using RC circuit attached to break

diff --git a/espcomm/espcomm_boards.c b/espcomm/espcomm_boards.c
index bb5cc26..c1da780 100644
--- a/espcomm/espcomm_boards.c
+++ b/espcomm/espcomm_boards.c
@@ -63,14 +63,16 @@ void espcomm_board_reset_into_app(espcomm_board_t* board)
 //

 // "ck" board: dtr pulls down gpio0, rts pulls down reset
+// also supports reset with RC circuit triggered by break signal

 void board_ck_rb()
 {
     serialport_set_rts(1);
     serialport_set_dtr(1);
+    serialport_send_break();
     espcomm_delay_ms(5);
     serialport_set_rts(0);
-    espcomm_delay_ms(50);
+    espcomm_delay_ms(250);          // wait for break to finish
     serialport_set_dtr(0);
 }

nerdralph, 2015-04-06:
p.s. the other reason why this is a big convenience is that most USB-TTL modules don't have both DTR and RTS broken out; usually just DTR.  With the auto-reset using the Tx line, DTR can be used to control GPIO0 for entering the bootloader.
The mod can be done right on the ESP-01 board; I soldered a 0805 cap between RST and Vcc, and a 0805 resistor diagonally between RST and the Rx pin.

igrr, 2015-04-06:
Isn't this the same as "wifio" reset method which is already implemented?

nerdralph, 2015-04-06:
I thought for wifio break controls GPIO0 through a PNP.  Will double-check the code though.

nerdralph, 2015-04-06:
Just looked at board_wifio_rb, and it toggles DTR to do the reset, and then sends a break to bring GPIO0 low:
    serialport_set_dtr(0);
    espcomm_delay_ms(5);
    serialport_set_dtr(1);
    espcomm_delay_ms(5);
    serialport_set_dtr(0);
    serialport_send_break();

compare that to my modified board_ck_rb:
    serialport_set_rts(1);
    serialport_set_dtr(1);
    serialport_send_break();
    espcomm_delay_ms(5);
    serialport_set_rts(0);
    espcomm_delay_ms(250);          // wait for break to finish
    serialport_set_dtr(0);

igrr, 2015-04-06:
Ok, got that, thanks.
Just one thing — your diff is not renederd correctly, could you please send it as a pull request, or surround the diff with code tags?

nerdralph, 2015-04-06:
I don't like forking a repo for a one-off commit - github doesn't need more dead forks.  Took me a couple minutes to find how to do code fencing, so here it is:
```
commit e2a0321446d301cdf4afb1c91327211e25afa1d8
Author: unknown <ralphdoncaster@gmail>
Date:   Sun Apr 5 22:51:30 2015 -0300

    add auto-reset using RC circuit attached to break

diff --git a/espcomm/espcomm_boards.c b/espcomm/espcomm_boards.c
index bb5cc26..c1da780 100644
--- a/espcomm/espcomm_boards.c
+++ b/espcomm/espcomm_boards.c
@@ -63,14 +63,16 @@ void espcomm_board_reset_into_app(espcomm_board_t* board)
 //

 // "ck" board: dtr pulls down gpio0, rts pulls down reset
+// also supports reset with RC circuit triggered by break signal

 void board_ck_rb()
 {
     serialport_set_rts(1);
     serialport_set_dtr(1);
+    serialport_send_break();
     espcomm_delay_ms(5);
     serialport_set_rts(0);
-    espcomm_delay_ms(50);
+    espcomm_delay_ms(250);          // wait for break to finish
     serialport_set_dtr(0);
 }
```

nerdralph, 2015-04-06:
p.p.s.  The upload tests I've been doing are at 921,600.  Part of the reason I'm confident the cap on RST is the reason for the stability is because it was so sensitive before I added the cap.  Simply touching a probe from my multimeter to the RST pin would usually reset the module, even when I tried adding a 15K pullup resistor to Vcc.  Now when I touch a probe to the pin, it doesn't reset the module.

commit 819b604a (2015-07-22): Auto-reset using an RC filter between RX and RST (by @nerdralph)

fix #4
closed 2015-07-22 by igrr