igrr/esptool-ck #5: Nodemcu board

mtornblad, 2015-04-08:
I have tried to use the Arduino IDE for ESP8266 but fails to upload the sketch after compile! I also tried to upgrade the esptool.exe to later version! I am running on Windows 8.1 (Version 6.3.9600) and my board is a Nodemcu board and it identifies as "USB-SERIAL CH340 (COM13)"

esptool.exe -vvvv -cd none -cb 115200 -cp COM13 -ca 0x00000 -cf sketch_apr08a.cpp_00000.bin -ca 0x40000 -cf sketch_apr08a.cpp_40000.bin 

esptool v0.4.2-4-ga0dd57d - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to none
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM13
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat C:\Users\Mattias\AppData\Local\Temp\build7182115499530350427.tmp/sketch_apr08a.cpp_00000.bin success
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
		flushed 0 bytes
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	read 0, requested 1
		espcomm cmd: didn't receive command response
trying to connect
	flush start
	setting serial port timeouts to 1 ms
		flushed 0 bytes
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	read 0, requested 1
		espcomm cmd: didn't receive command response
trying to connect
	flush start
	setting serial port timeouts to 1 ms
		flushed 0 bytes
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	serialport_receive_C0: 73 instead of C0
		espcomm cmd: didn't receive command response
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
		flushed 0 bytes
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	read 0, requested 1
		espcomm cmd: didn't receive command response
trying to connect
	flush start
	setting serial port timeouts to 1 ms
		flushed 0 bytes
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	serialport_receive_C0: 73 instead of C0
		espcomm cmd: didn't receive command response
trying to connect
	flush start
	setting serial port timeouts to 1 ms
		flushed 2 bytes
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	read 0, requested 1
		espcomm cmd: didn't receive command response
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
		flushed 0 bytes
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	read 0, requested 1
		espcomm cmd: didn't receive command response
trying to connect
	flush start
	setting serial port timeouts to 1 ms
		flushed 0 bytes
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	read 0, requested 1
		espcomm cmd: didn't receive command response
trying to connect
	flush start
	setting serial port timeouts to 1 ms
		flushed 0 bytes
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	read 0, requested 1
		espcomm cmd: didn't receive command response
warning: espcomm_sync failed
error: espcomm_open failed

goseese, 2015-04-08:
Are you sure you have good serial comm to the board? 

You can try to open serial monitor to that port in your Arduino IDE at 115200. Then ground GPIO0 and toggle the reset pin. You should see at least some garble in the serial monitor. If you don't see anything perhaps your TX/RX are swapped or there is some other serial comm issue.

mtornblad, 2015-04-08:
Yes,

If i start the board in upload mode i can see garbage in the serial console, and the nodemcu gui tools works to flash the board!

goseese, 2015-04-08:
I was trouble shooting a problem like this once and what helped me was to put a 2nd FTDI adaptor's RX pin on the TX of the ESP8266. Then I opened a screen to that 2nd serial port and watched the return data from the ESP8266. If you can do that you might be able to see if the ESP8266 is indeed returning a 73 instead of a C0. It almost sounds like a baud rate issue, the EPS8266 might not be responding at 115200. What baud rate is your nodemcu gui tool working at? I am on a unix system and have never used a nodeMCU gui.

mtornblad, 2015-04-09:
When connecting a second FTDI i saw that the chip was rebooted during the update. I changed the command to use board "CK" and now it is working! It manages to switch from runmode to fw upload mode automatic! That's Gr8! ;-)

https://raw.githubusercontent.com/nodemcu/nodemcu-devkit/master/Documents/NODEMCU_DEVKIT_SCH.png

closed 2016-05-10 by igrr

VRuzhentsov, 2019-04-12:
In my case I fixed this error setting up correct board. Mine was `nodemcu` instead of `ck`