igrr/esptool-ck #9: upload problem encountered in Arduino IDE

boneskull, 2015-05-26:
Here's the end of trying to build `WiFiScan` example using the `1.6.4-673-g8cd3697` board definition:

```
/Users/boneskull/Library/Arduino15/packages/esp8266/tools/esptool/0.4.4/esptool -eo 
/var/folders/d4/vrj66_rs375gh9_33xr9bnv80000gs/T/build7455700842258194699.tmp/WiFiScan.cpp.elf 
-bo /var/folders/d4/vrj66_rs375gh9_33xr9bnv80000gs/T/build7455700842258194699.tmp/WiFiScan.cpp_00000.bin 
-bm qio -bf 40 -bz 512K -bs .text -bs .data -bs .rodata -bc -ec -eo 
/var/folders/d4/vrj66_rs375gh9_33xr9bnv80000gs/T/build7455700842258194699.tmp/WiFiScan.cpp.elf 
-es .irom0.text 
/var/folders/d4/vrj66_rs375gh9_33xr9bnv80000gs/T/build7455700842258194699.tmp/WiFiScan.cpp_10000.bin 
-ec 

Sketch uses 231,360 bytes (44%) of program storage space. Maximum is 524,288 bytes.
Uploading 32848 bytes from /var/folders/d4/vrj66_rs375gh9_33xr9bnv80000gs/T/build7455700842258194699.tmp/WiFiScan.cpp_00000.bin to flash at 0x00000000
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
```

Trying the upload manually, with a somewhat different command, also fails:

```
$ esptool -cp /dev/cu.usbserial -cf WiFiScan.cpp_00000.bin -ca 0x10000 -cf WiFiScan.cpp_10000.bin -vvvv
esptool v0.4.4 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting port from /dev/tty.usbserial to /dev/cu.usbserial
	espcomm_upload_file
	stat WiFiScan.cpp_00000.bin success
opening port /dev/cu.usbserial at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
		espcomm_send_command: didn't receive command response
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
		espcomm_send_command: received 2 bytes:
		0x00
		0x00
		espcomm_send_command: response 0x20120707
	espcomm_send_command: receiving 2 bytes of data
		espcomm_send_command: received 2 bytes:
		0x00
		0x00
		espcomm_send_command: response 0x20120707
	espcomm_send_command: receiving 2 bytes of data
		espcomm_send_command: received 2 bytes:
		0x00
		0x00
		espcomm_send_command: response 0x20120707
	espcomm_send_command: receiving 2 bytes of data
		espcomm_send_command: received 2 bytes:
		0x00
		0x00
		espcomm_send_command: response 0x20120707
	espcomm_send_command: receiving 2 bytes of data
		espcomm_send_command: received 2 bytes:
		0x00
		0x00
		espcomm_send_command: response 0x20120707
	espcomm_send_command: receiving 2 bytes of data
		espcomm_send_command: received 2 bytes:
		0x00
		0x00
		espcomm_send_command: response 0x20120707
	espcomm_send_command: receiving 2 bytes of data
		espcomm_send_command: received 2 bytes:
		0x00
		0x00
		espcomm_send_command: response 0x20120707
	espcomm_send_command: receiving 2 bytes of data
		espcomm_send_command: received 2 bytes:
		0x00
		0x00
		espcomm_send_command: response 0x20120707
	espcomm_open
Uploading 32848 bytes from WiFiScan.cpp_00000.bin to flash at 0x00000000
	erasing flash
	size: 008050 address: 000000
	first_sector_index: 0
	total_sector_count: 9
	head_sector_count: 9
	adjusted_sector_count: 5
	adjusted_size: 005000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 10000
	setting character timeout 100
	done
	setting timeout 1
	setting character timeout 1
	done
	espcomm_send_command: receiving 1796 bytes of data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader
```

I tried various combinations of flags `-bf`, `-bm` and `-cb` to no avail.

But `esptool.py` uploads fine:

```
$ esptool.py --port /dev/cu.usbserial write_flash 0x00000 WiFiScan.cpp_00000.bin 0x10000 WiFiScan.cpp_10000.bin
Connecting...
Erasing flash...
Writing at 0x00008000... (100 %)
Erasing flash...
Writing at 0x00040400... (100 %)

Leaving...
```

Setup:

- esptool v0.4.4
- ESP-01
- PL-2303HX USB-to-serial adapter [like this](http://www.amazon.com/Kootek-PL2303-Serial-Converter-Adapter/dp/B00FEAMUOK/ref=sr_1_3?s=electronics&ie=UTF8&qid=1432623776&sr=1-3&keywords=pl2303hx)
- Mac OS 10.10.3
- pretty sure I'm using the FTDI v2.3 driver [as found here](http://www.ftdichip.com/Drivers/VCP.htm) but I don't know how to determine that.

igrr, 2015-05-26:
```
    espcomm_send_command: receiving 1796 bytes of data

igrr, 2015-05-26:
Investigated this a bit. Totally weird.
The issue comes from the fact that the reply packet has incorrect size or checksum.
I tried to print out each received byte:
```
Uploading 31680 bytes from /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build2638420307500603799.tmp/sketch_may25a.cpp_00000.bin to flash at 0x00000000
	erasing flash
	size: 007bc0 address: 000000
	first_sector_index: 0
	total_sector_count: 8
	head_sector_count: 8
	adjusted_sector_count: 4
	adjusted_size: 004000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
		serialport_get_timeout: 1000
		serialport_set_timeout: 10000
	setting timeout 10000
	setting character timeout 100
	done
		old_timeout = 1000
c0 
		serialport_set_timeout: 1000
	setting timeout 1000
	setting character timeout 10
	done
01 
02 
02 
00 
07 
90 
02 
40 
		rp.direction = 1
		rp.command = 2
		rp.size = 2
		rp.checksum = 40029007
	espcomm_send_command: receiving 2 bytes of data
08 
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader
```
Here the checksum is already broken (on other runs size was broken as well).

Now when I connect a scope, I see the ESP is sending the correct data:
[image]

But the bytes I read from the port are different. Looking further.

briankimstudio, 2015-06-09:
Hi, I had same problem on Mac OS X Yosemite 10.10.3 and it is fixed by esptool.py. 
Short tutorial for switching uploader from esptool to esptool.py on Mac

http://hpclab.blogspot.com/2015/06/esp8266-arduino-ide-on-mac-os-x.html

Hope it would help someone who just jumped into ESP8266 world

stblassitude, 2015-08-19:
It might be worthwhile adding a note to the README, since PL230x based adapters are quite widespread.

igrr, 2015-08-19:
@stblassitude I've added a note to Arduino docs.

boneskull, 2015-08-19:
FWIW I've since found the CP2102 devices are the most reliable when working with ESP8266's

etiquet, 2016-04-23:
Hello gentlemen, I was having constantly the same issue with lolin nodemcu 1.0 HW that uses CH340 chip.

I apply the following workaround : http://arduino.stackexchange.com/questions/20219/upload-with-esptool-fails-with-espcomm-send-command-cant-receive-slip-payload installing the esptool.py that works fine for me. (file is platform.txt instead of hardware.txt described)

so... practically .... attached the files that works for me now :
(Arduino 1.6.8 /  ESP8266 2.20 / OS X 10.11.4 )

[platform.txt](https://github.com/igrr/esptool-ck/files/232935/platform.txt) 
location >> ~/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0
[esptool.py.zip](https://github.com/igrr/esptool-ck/files/232936/esptool.py.zip)
location >>~/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/tools
(IDE to be restarted)

Dear Ivan, why not considering including this in the official package for Mac with an instruction somewhere ?

samyk, 2016-10-12:
I attempted the update above from @etiquet on 2.2.0 with no luck, error being ". Note I kept GPIO 0 low via 10K ohm resistor when plugging board in using a common programmer (http://samy.pl/o/espp.jpg). Error is "Failed to connect to ESP8266". I've tried both esptool 1.1 and 1.2-dev.

Same when attempting from console:
$ sudo esptool.py --port /dev/cu.Repleo-CH341-00005014 write_flash 0x00000 build5f5ea63a04bf75fd9ebf7eb6ec082a45.tmp/Blink.ino.bin 
esptool.py v1.2-dev
Connecting...

A fatal error occurred: Failed to connect to ESP8266

rousnay, 2017-01-16:
Try to connect The GPIO0 with a 10k resistor to ground and the CH_PD to VCC and it works :)