igrr/esptool-ck #10: Can't connect to my NodeMCU! Is it dead?

jeannot9, 2015-05-28:
After over 12 hours of trying to make my NodeMCU work, I am being overcome by desperation.
Any idea, or opinion that can help me get this thing talking is welcome.

Windows 7 64-bits
NodeMCU DevKit 0.9

As recommended, I started with installing the CH340 USB Serial driver with the NodeMCU unplugged. I found a 2011 version of the driver, and a 2014 version. 

Installing WCH.CN
    |__ USB-SERIAL CH340
        |__ 11/04/2011, 3.3.2011.11

After the driver installs, Device Manager shows me NO new COM port, which I think is the first bad sign. And I get the message:

"The drive is successfully Pre-installed in advance!" Whatever that means...

Of course, ESP8266Flasher cannot see a new COM port either.

So I went after this as a driver problem. After an Uninstall, the next install (2011, or 2014) said some files were already present. So the uninstalls leave some duty behind...

I then manually cleaned up the registry, any file on disk starting with "CH34", ran malwarebytes to fix inconsistencies and retried. To no avail.

Sooo, at this point, with a driver installed:

-When I connect the NodeMCU to my USB port, I hear the Windows Plug&Play sound, like when we connect a mouse or anything else. And the Device Manager refreshes, so something is happening. 

Windows indicates "Installing device driver software..." for a few seconds.
then:
"Device driver software was not successfuly installed"

-When the NodeMCU is connected, the blue LED flashes 4-5 times and a "ESP_A265C8" thing appears on my phone as a WiFi accessible access point. (My phone tells me "54Mbps", 2.4GHz, Connected and Signal Excellent)

Enough seems to be working to drive me bezerk.

The same thing happens on my laptop running XP, and another laptop running Windows 8. Which points to either a driver or a hardware problem. Or a head problem.

But all over the place, I hear stories of seamless installs.

I tried resetting the NodeMCU by shorting the RST pin to ground for a few seconds. Same results.

I have another NodeMCU on order.

Q1: Should I see a new COM port in Device Manager after installing the driver (and before plugging the NodeMCU)?

Q2: Has anyone experienced this and concluded that either the driver was bad, wrongly installed or a bad NodeMCU?

Thank for your help!!!

jeannot9, 2015-06-06:
Issue resolved: two, 2, I repeat, two bad USB cables.

closed 2015-07-22 by igrr

iuliancb, 2015-11-06:
Just does not work ! bad design

iuliancb, 2015-11-06:
mine does not even show up in the COM ports no matter how hard i try.Have tried different cables on different windows versions and still no luck ! Seems like the board is busted !

iuliancb, 2015-11-06:
A piece of crap you woud rather get just the module itself and do not bother with the nodeMCU dev kit !