igrr/esptool-ck #16: Add delay after reset option

vicatcu, 2015-10-21:
I need a new 'feature' (I think it doesn't already exist) to support a board I'm working on. It's quite simple. After doing the RTS and DTR control logic, I might need to inject a fixed delay before engaging in Serial download. The reason is because in my design USB serial traffic can only reach the ESP8266 by being repeated through another microcontroller. That microcontroller itself has a bootloader, and it is *also* reset using the DTR line. So I need to clear the timeout for the upstream microcontroller bootloader before sending serial data destined for the ESP8266 (through it). Does that makes sense?

Reading through the code, I think the delay would probably need to be inserted right before [here](https://github.com/igrr/esptool-ck/blob/6de5ee55e1efac4ae1dfb972143d288fb75715e5/espcomm/espcomm.c#L235) / after [here](https://github.com/igrr/esptool-ck/blob/6de5ee55e1efac4ae1dfb972143d288fb75715e5/espcomm/espcomm.c#L234).

I can think of two ways to integrate the feature. The first would be to expose a new field (e.g. post_reset_delay) in the [espcomm_board_](https://github.com/igrr/esptool-ck/blob/6de5ee55e1efac4ae1dfb972143d288fb75715e5/espcomm/espcomm_boards.c#L7-L12) structure, with all the existing board definitions set to 0 or something. The second way would be to expose it as a new arg_parse handler, probably a new case [here](https://github.com/igrr/esptool-ck/blob/6de5ee55e1efac4ae1dfb972143d288fb75715e5/argparse/argparse_commcmd.c#L43) (e.g. -cr). I don't know that one of these is better than the other philosophically.

@igrr What do you think? If it's acceptable and of general interest, and you have a can provide guidance for which way to go about implementation, I would be happy to implement it and submit a pull request. FWIW, I'm planning to implement a very similar reset circuit to the nodemcu one.

vicatcu, 2015-10-22:
After chatting with @me-no-dev in the gitter, it may be more elegant to handle this in my bootloader. That is to discriminate the serial traffic received by my bootloader as destined for the microcontroller or destined for pass-through to the ESP8266. In any case, I still wonder if this might be a useful feature to implement?

igrr, 2015-10-29:
I think adding an argument and a corresponding arg_parse handler would be a good option.

At some point i was thinking about adding support for "scripts" which define how reset should be performed. Currently there is some C code which defines different reset methods. This works fine, but isn't very flexible. So the idea was to handle 'reset method' arguments like
```
-cb dtr:1,rts:1,delay:5,rts:0,delay:30,dtr:0
```
This way anyone could use esptool-ck with custom reset circuit, without having to modify C code.
I think this feature would work for your case as well. Does this make sense?

kodehnal, 2016-02-25:
Did you implemented the  'reset method' arguments already? I have this simple reset and GPIO0 insert circuit at my board http://www.ba0sh1.com/wp-content/uploads/2015/01/ESP8266-boot-select-sch.png.
DTR line is connected paralel to the reset switch. But the DTR low signal must be inserted at least for 500mS to charge the capacitor and keep the GPIO0 low when the reset conditions occured.