igrr/esptool-ck #20: Mac OS X El Capitan does not work

DiploGrand, 2015-11-22:
When I used it previously, it's work fine, but after update on El Capitan have an error:
`warning: espcomm_sync failed`
`error: espcomm_open failed`

P.S. On virtual machine with Windows 7 work fine.

Pavlentiycomua, 2015-11-22:
I have the same error, in VM ok, but in VM is uncomfortable work. 
Do you have a solution?

igrr, 2015-11-22:
Running El Cap here, no issues. Which Serial adapter chip are you guys using? Also, which driver?

Pavlentiycomua, 2015-11-22:
I'm using original arduino uno board for flashing esp8266

DiploGrand, 2015-11-22:
I tried on the Arduino UNO with standard drivers from the Arduino IDE, I also tried it with USB-UART CP2102 with driver SiLabs VCP. On Windows, both versions work fine.

DiploGrand, 2015-11-22:
Here, too, discussing:
http://www.esp8266.com/viewtopic.php?f=26&t=5831

Pavlentiycomua, 2015-11-23:
I took Sparkfun cp2102 from a friend and it work fine, but still don't understand why uno board not working in macOS, but well working in windows.

igrr, 2015-11-23:
Good to know that CP2102 works fine for you (that's what I'm using myself with El Cap).

I'm not familiar with Uno, which USB-serial chip does it use? I suppose that OS X driver might be handling timeouts or RTS/DTR pin manipulation differently.
Could please you also mention what kind of reset circuit do you use? NodeMCU-like or something like RTS/DTR pins connected to GPIO0/CHPD?

Pavlentiycomua, 2015-11-23:
Work fine in Win by this scheme http://www.stefanocottafavi.com/discover-the-esp8266/ 
Usb-serial chip on Uno R3 is Atmega16U2 and I have an idea to update firmware on this chip. Take me some time to try updating firmware.

Pavlentiycomua, 2015-11-23:
Sorry, on Uno usb-serial chip is Atmega8U2

Pavlentiycomua, 2015-11-23:
Turns my chip has latest firmware and no need to update it, so problem is still need to resolve.

awootton, 2015-11-25:
I'm very stuck and I'm about to have to build and debug this code. I hope someone can help.
I am using the signed version of the osx drivers (repleo) and it appears to work in the sense that a nodemcu 0.9 that I previously programmed can output to serial. I have had success with the same mac when with ubuntu. Anyway, here's the error (which I can reproduce in the command line) 

Sketch uses 207,312 bytes (19%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 44,812 bytes (54%) of dynamic memory, leaving 37,108 bytes for local variables. Maximum is 81,920 bytes.
/Users/awootton/Library/Arduino15/packages/esp8266/tools/esptool/0.4.5/esptool -vv -cd ck -cb 115200 -cp /dev/cu.Repleo-CH341-00002014 -ca 0x00000 -cf /var/folders/cw/2nkqgg7j2_5__dhb1_2xh3dw0000gn/T/build948846584884931390.tmp/NodeMcuBinkAtwTest1.cpp.bin 
esptool v0.4.5 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.Repleo-CH341-00002014
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/cw/2nkqgg7j2_5__dhb1_2xh3dw0000gn/T/build948846584884931390.tmp/NodeMcuBinkAtwTest1.cpp.bin success
opening port /dev/cu.Repleo-CH341-00002014 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed

teomaragakis, 2015-12-11:
@igrr  how did you manage to make CP2102 work under El Capitan? This driver is killing me.

igrr, 2015-12-13:
@teomaragakis Removed all 3rd party kexts which were left over from my Yosemite installation, then installed Silabs VCP driver. Pretty much didn't do anything else.
(I had to remove 3rd party kexts because my OS upgrade process got stuck. Turned out it was due to Virtualbox kext installed...)

teomaragakis, 2015-12-14:
@igrr I think it's my adaptor then. Thanks!

boneskull, 2016-03-01:
Hi,

I am having perhaps a related issue.  I'm using PlatformIO to flash my D1 Mini.  This is the upload log:

```
/Applications/CLion.app/Contents/bin/cmake/bin/cmake --build /Users/boneskull/Library/Caches/CLion12/cmake/generated/549bbd3d/549bbd3d/Debug --target PLATFORMIO_UPLOAD -- -j 8
[Tue Mar  1 10:45:54 2016] Processing d1_mini (platform: espressif, lib_install: 551, 13, 25, upload_speed: 230400, board: d1_mini, framework: arduino)
--------------------------------------------------------------------------------
<lambda>(["upload"], [".pioenvs/d1_mini/firmware.bin"])
Auto-detected UPLOAD_PORT/DISK: /dev/cu.wchusbserial141110
"/Users/boneskull/.platformio/packages/tool-esptool/esptool" -vv -cd nodemcu -cb 230400 -cp /dev/cu.wchusbserial141110 -cf .pioenvs/d1_mini/firmware.bin
esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting board to nodemcu
setting baudrate from 115200 to 230400
setting port from /dev/tty.usbserial to /dev/cu.wchusbserial141110
espcomm_upload_file
espcomm_upload_mem
opening port /dev/cu.wchusbserial141110 at 230400
tcgetattr
tcsetattr
serial open
opening bootloader
resetting board
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 2 bytes of data
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 2 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x01 0x07, expected 0x01 0x08
resetting board
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 2 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x01 0x07, expected 0x01 0x08
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 2 bytes of data
resetting board
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 2 bytes of data
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 2 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x01 0x07, expected 0x01 0x08
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
scons: *** [upload] Error 255
========================= [ ERROR ] Took 2.88 seconds =========================
make[3]: *** [CMakeFiles/PLATFORMIO_UPLOAD] Error 1
make[2]: *** [CMakeFiles/PLATFORMIO_UPLOAD.dir/all] Error 2
make[1]: *** [CMakeFiles/PLATFORMIO_UPLOAD.dir/rule] Error 2
make: *** [PLATFORMIO_UPLOAD] Error 2
```

I'm unsure if this is a misconfiguration on PlatformIO's part, or if it's an esptool problem.  Perhaps `nodemcu` is not the right "board"; a NodeMCU board uses a CP2102 adapter if I'm not mistaken, and the D1 mini uses a CH340G.  Also maybe it needs to be in a different DIO/QIO mode?

`esptool.py` will upload the `.bin` file successfully:

```
$ esptool.py --baud 230400 --port /dev/tty.wchusbserial141110 write_flash 0x00000 firmware.bin
Connecting...
Erasing flash...
Wrote 242688 bytes at 0x00000000 in 17.7 seconds (109.6 kbit/s)...

Leaving...
```

igrr, 2016-03-01:
@boneskull are you using a PL2303 based adapter? These aren't currently supported by esptool-ck on OS X, please see https://github.com/igrr/esptool-ck/issues/9.

boneskull, 2016-03-01:
@igrr No, the D1 mini has a built-in CH340G adapter

igrr, 2016-03-02:
@boneskull ah, right. I have one CH340G device — NodeMCU 0.9. Just tried that on OS X 10.11.3 with the driver downloaded from https://tzapu.com/making-ch340-ch341-serial-adapters-work-under-el-capitan-os-x/. Didn't work initially, but then I swapped USB cables and then it worked 10 out of 10 times at 230400 baud.

```
/Users/igrokhotkov/projects/esp8266/arduino.cc/master/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/esptool/esptool -vv -cd nodemcu -cb 230400 -cp /dev/cu.wchusbserial1420 -ca 0x00000 -cf /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build931f29da8146a91a28f7b4341bdae31e.tmp/Blink.ino.bin 
esptool v0.4.7 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 230400
	setting port from /dev/tty.usbserial to /dev/cu.wchusbserial1420
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.wchusbserial1420 at 230400
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 235376 bytes from to flash at 0x00000000
	erasing flash
	size: 039770 address: 000000
	first_sector_index: 0
	total_sector_count: 58
	head_sector_count: 16
	adjusted_sector_count: 42
	erase_size: 02a000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 15000
	setting character timeout 150
	done
	setting timeout 1
	setting character timeout 1
	done
	espcomm_send_command: receiving 2 bytes of data
	writing flash
......................................................................................................................................................................................................................................
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader

```

boneskull, 2016-03-02:
Will try a diff cable

igrr, 2016-03-02:
After a few hours of flashing finally got this:
```
warning: espcomm_send_command: wrong direction/command: 0x01 0x07, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x01 0x07, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x01 0x07, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x01 0x07, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x01 0x07, expected 0x01 0x08
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
```
So I can confirm this happens, but i can't reproduce it easily enough to debug this.

boneskull, 2016-04-12:
I am having somewhat reliable results using 57600 baud, unfortunately

peteroid, 2016-05-06:
Sometimes I encountered this issue. It could be fixed by unplugging some pins off before flashing. Seems like some issue about the UART pins?

fah, 2016-05-15:
Did you check the supply voltage? 

I'm using a plain usb serial converter CP2101 and connected the ESP8266-01 right to the **5V** supply and data pins. ESP gets warm but works. 
There is a 220uF across the 5V supply.
Also the supply source needs to be checked over time when uploading. (Oscilloscope showed a -500mV variation (this is -10%!) in the supply when uploading.)
Maybe with 3.3V it works more stable on high speed?
This maybe explains why a different USB cable works (thicker cable?).

For me 115200 baud works well to do some AT commands. But neither Arduino 1.6.8 IDE nor platformio are able to upload the wifiscanner,ino example.

Below are log outputs with different baud rates 115200,9600,230400.

115200baud:
Arduino IDE:
```
opening port /dev/cu.SLAB_USBtoUART at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2013 bytes of data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
...
```

esptool,py/platformIO:
```
$ /usr/local/bin/esptool.py --port /dev/cu.SLAB_USBtoUART --baud 115200 write_flash 0x00000 .pioenvs/esp01/firmware.bin
esptool.py v1.0.2-dev
Connecting...

A fatal error occurred: Failed to connect to ESP8266
```

-------

Trying with 9600 baud:

```
/Users/x/Library/Arduino15/packages/esp8266/tools/esptool/0.4.8/esptool -vv -cd ck -cb 9600 -cp /dev/cu.SLAB_USBtoUART -ca 0x00000 -cf /var/folders/......./WiFiScan.ino.bin 
esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 9600
	setting port from /dev/tty.usbserial to /dev/cu.SLAB_USBtoUART
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.SLAB_USBtoUART at 9600
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
...
...
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
```
-------

Trying with 230400 baud:

```
Users/x/Library/Arduino15/packages/esp8266/tools/esptool/0.4.8/esptool -vv -cd ck -cb 230400 -cp /dev/cu.SLAB_USBtoUART -ca 0x00000 -cf /var/folders/........../WiFiScan.ino.bin 
esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 230400
	setting port from /dev/tty.usbserial to /dev/cu.SLAB_USBtoUART
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.SLAB_USBtoUART at 230400
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
...

...
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: E0 instead of C0
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
```

MAC OS X 10.11.4
CP2101 (www.betemcu.cn, BTE13-007)
ESP8266-01 (AI-cloud inside, 2 led's, 2x512kB)

fah, 2016-05-15:
**Problem solved here**:
1. connect esp VCC to 3.3V
2. 3k3 resistor between CH_PD and VCC
3. connect GND and PD0 for programming
4. 220uF between GND and VCC

Result:
´´´
esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.SLAB_USBtoUART
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.SLAB_USBtoUART at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 228128 bytes from to flash at 0x00000000
	erasing flash
	size: 037b20 address: 000000
	first_sector_index: 0
	total_sector_count: 56
	head_sector_count: 16
	adjusted_sector_count: 40
	erase_size: 028000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 15000
	setting character timeout 150
	done
	setting timeout 1
	setting character timeout 1
	done
	espcomm_send_command: receiving 2 bytes of data
	writing flash
...............................................................................................................................................................................................................................
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader
´´´

Immediately the serial monitor showed the wifi scan result. So everything fine here.