igrr/esptool-ck #22: Esptool 4.6  not reliable on my flash setup.

far5893, 2016-01-11:
I try differend baudrate  but not seem rs232 related

/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/../../../tools/esptool/0.4.6/esptool -vv -cd none  -cb 460800  -cp /dev/ttyUSB0 -ca 0x100000 -cf /tmp/build8924734518208192267.tmp/esp8266-WebServices.spiffs.bin 
esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to none
	setting baudrate from 115200 to 460800
	setting port from /dev/ttyUSB0 to /dev/ttyUSB0
	setting address from 0x00000000 to 0x00100000
	espcomm_upload_file
	stat /tmp/build8924734518208192267.tmp/esp8266-WebServices.spiffs.bin success
opening port /dev/ttyUSB0 at 460800
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_open
Uploading 3125248 bytes from /tmp/build8924734518208192267.tmp/esp8266-WebServices.spiffs.bin to flash at 0x00100000
	erasing flash
	size: 2fb000 address: 100000
	first_sector_index: 256
	total_sector_count: 763
	head_sector_count: 16
	adjusted_sector_count: 747
	adjusted_size: 2eb000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 10000
	setting character timeout 100
	done
	setting timeout 1
	setting character timeout 1
	done
	espcomm_send_command: receiving 2 bytes of data
	writing flash
................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: didn't receive command response
closing bootloader

but sometimes it stops immediately with FLASH_DOWNLOAD_BEGIN

I suppose that my flash is slow and some timeout occur during writing data or starting writing,
sketck and flash work well , only spiffs 3M  stop randomly upload.

Links2004, 2016-01-11:
have you tried using less then 460800 baud?

igrr, 2016-01-11:
Failure at FLASH_DOWNLOAD_BEGIN stage usually has to do with flash erase time exceeding the 10 second timeout. This happens when you try to upload a large (3M) spiffs binary.

If it fails somewhere in the middle of the process, then it might indicate communication errors.
Capturing serial data on both TX and RX lines using a logic analyzer might help gain some insight into this type of issue.

far5893, 2016-01-11:
it would be possible to split esptool in two stage : flash erase , and programming
with the possibility to restart/recover from last good block (with only this block erase if necessary )?
Transfer 3Mb of data without any error correction/recovery its very hard even with perfect serial adapter, shot wires , and no software bug!!!