igrr/esptool-ck #23: Unable to pull GPIO0 down

jdkanani, 2016-01-21:
I am not sure what causing this issue. But, flashing with native SDK firmware works well. First time I used arduino compiled binary (using esptool-ck to flash via raspberrypi) causes this issue:

```
esptool-ck/esptool -vv -cd none -cb 115200 -cp /dev/ttyAMA0 -ca 0x00000 -cf main.ino.generic.bin
```

I am using board `none` - not sure if it has anything to do with that. More info about this problem: https://github.com/esp8266/Arduino/issues/1474

@igrr any idea?

igrr, 2016-01-21:
I think esptool doesn't (shouldn't) care what kind of binary (compiled with Arduino of IoT SDK) it flashes.
If you are using `-cd none`, it will not manipulate DTR and RTS pins.
If you are connecting DTR/RTS pins to GPIO0, and want GPIO0 to be pulled low automatically, please use `-cd ck` or `-cd nodemcu`.
It would be cool if you could list the steps to reproduce the issue for me, so that I may be able to help you.

jdkanani, 2016-01-21:
Here is the code I used to export binary from Esp8266 Arduino:

```arduino
//
// main.ino
//
void setup() {
  pinMode(12, OUTPUT); 
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(12, LOW); 
  delay(1000); 
  digitalWrite(12, HIGH);
  delay(2000);
}
```
`Export compiled binary` option compiles and generates `main.ino.generic.bin`. I copied that bin file to my raspberry pi which is connected to ESP12 with following connections:

ESP12 | RaspberryPI
------ | -----------------
VCC, CH_PD | 3.3v
GND, GPIO15 | GND
TX | RX
RX | TX
GPIO0 | GND (To flash new firmware)

`GPIO12` is connected to LED to check blink.

To flash that bin file I used following command:
```
esptool-ck/esptool -vv -cd none -cb 115200 -cp /dev/ttyAMA0 -ca 0x00000 -cf main.ino.generic.bin
```

While floating `GPIO0`, Blink example works fine. But, grounding `GPIO0` again (second time, to flash new firmware) causes a hang in RaspberryPi.

Native SDK compiled binary works fine using this wiring.