igrr/esptool-ck #27: Chip Erase crashes program on WinXP and NodeMCU 0.9 board

furyfire, 2016-03-01:
Running 
`esptool.exe -cp COM9 -cd nodemcu -ce`
results 
`Unhandled exception at 0x004013f2 in esptool.exe: 0xC0000005: Access violation reading location 0x00000000.`

igrr, 2016-03-01:
Does it happen with esptool 0.4.8 or 0.4.7?
Initially 2.1.0 version of esp8266 core was distributed with 0.4.7, but a bug in esptool was fixed later on, now it ships with 0.4.8. Could you please run `esptool -h` and check?

furyfire, 2016-03-01:
C:\Documents and Settings\JCT\.platformio\packages\tool-esptool>esptool.exe -h
esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>

furyfire, 2016-03-01:
Full log output from first a write then a chip erase

```
D:\Dev\ESP8266\Issue1706>D:\esptool.exe -vv -cd nodemcu -cb 115200 -cp COM9 -cf .pioenvs\nodemcu\firmware.bin
esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting board to nodemcu
        setting baudrate from 115200 to 115200
        setting port from COM1 to COM9
        espcomm_upload_file
        espcomm_upload_mem
        setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        read 0, requested 1
trying to connect
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
Uploading 238080 bytes from to flash at 0x00000000
        erasing flash
        size: 03a200 address: 000000
        first_sector_index: 0
        total_sector_count: 59
        head_sector_count: 16
        adjusted_sector_count: 43
        erase_size: 02b000
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        setting serial port timeouts to 15000 ms
        setting serial port timeouts to 1000 ms
        espcomm_send_command: receiving 2 bytes of data
        writing flash
.........................................................................................................................................................................................................................................
starting app without reboot
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
closing bootloader
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete

D:\Dev\ESP8266\Issue1706>D:\esptool.exe -vv -cd nodemcu -cb 115200 -cp COM9 -ce
esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting board to nodemcu
        setting baudrate from 115200 to 115200
        setting port from COM1 to COM9
        espcomm_erase_flash
        setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        read 0, requested 1
trying to connect
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        setting serial port timeouts to 1000 ms
        setting serial port timeouts to 1000 ms
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data

D:\Dev\ESP8266\Issue1706>
```