igrr/esptool-ck #28: DTR/RTS getting manipulated even if reset method is set to none

emiliojazz, 2016-03-09:
Hello, I'm having a problem trying to upload a sketch onto ESP8266 using an Arduino MEGA as a usb serial adapter.
Please note that the MEGA is not merely used as a serial adapter, but it goes into that mode by issuing a specific command from the terminal. Then it goes to serial mirroring mode, and triggers the ESP8266 to go into bootloader mode. It should be ready to flash the ESP8266.
The problem is that when I start flashing from Arduino IDE (setting the resetmethod to none) the Arduino board gets resetted so I lose the serial mirroring connection, and the esptool rightly outputs espcomm_sync_failed, espcomm_open_failed stuff.
What am I missing? It should be fairly trivial to upload a sketch when the serial mirroring is activated and NO DTR/RTS intervention is set.
Thank you very much!
Emiliano

t3chguy, 2016-03-14:
I believe I am having the same issue (at least same symptoms) with a Genuino101 (which doesn't really care for DTR/RTS but it is still breaking things)

igrr, 2016-03-16:
When OS does this happen on?

emiliojazz, 2016-03-17:
@igrr I'm working on Linux Ubuntu 14.04 using Arduino IDE 1.6.6 and esptool 0.4.8

emiliojazz, 2016-03-21:
The culprit was the serial driver, it manipulates DTR/RTS every time an "open" on the serial port is called.
For now I found a workaround by increasing the timeout time in esptool and checking an hardware button at boot to decide if to go in mirroring mode or not.
Definitely not the easiest way to do it...but I didn't find a way to disable DTR/RTS manipulation by the driver so far (seems to be possible with Windows...strangely).
Thank you for your patience, and hope that this post would help someone stucking with the same problem.
Regards

igrr, 2016-03-21:
I came to the same conclusion. I'm able to disable DTR/RTS manipulation on OS X, but on Linux the driver handles things a bit differently.