igrr/esptool-ck #30: Problem with writing files that are not a multiple of the flash block size?

dbetz, 2016-03-11:
I've been trying to use esptool-ck instead of the Python version of esptool for loading files generated by the esp-link project. At first, the loaded code wouldn't work correctly. Then I noticed that the Python code was always loading a multiple of the flash block size (1024). I wrote a trivial program to pad all of the files I was trying to load to a multiple of 1024 bytes and now the code loaded by esptool-ck seems to work fine. Is there a bug in esptool-ck when trying to load files that are not a multiple of the flash block size?

dbetz, 2016-03-15:
I'm not sure I diagnosed this problem correctly because I've modified my image and now even padding to a 1024 byte boundary doesn't work. Does esptool-ck erase flash sectors before writing them or is it necessary to use the -ce option first to erase the flash? If so, what is the syntax for using -ce because I get errors when I put it in my command line ahead of the commands to write to the flash.

igrr, 2016-03-16:
Erasing sectors is part of the ESP bootloader protocol, and it is handled automatically by the ESP. I.e. esptool-ck tells ESP how much data it wants to upload, and the ESP erases the given number of sectors. Then esptool-ck goes on to transmit said amount of data to the ESP.

closed 2016-06-17 by igrr