igrr/esptool-ck #31: flash size is not working

ovaltineo, 2016-03-13:
I tried flashing Espruino using this tool but the flash size doesn't seem to have any effect on my ESP-07 and NODEMCU modules.  If I use -v, it says "setting flash size from 512K to 1M" or "setting flash size from 512K to 4M" but according to Espruino, the memory map is just 512K.

If I use the official ESP Flash Download Tool, it sets the flash size properly and Espruino reports the correct memory map.  But this tool is a pain because it doesn't use DTR/RTS and cannot be invoked via command line.

igrr, 2016-03-14:
Flash size parameter is not used when uploading the binary image. It is only used when the image is generated. So yes, this is expected.

ovaltineo, 2016-03-14:
That's what I thought, but I only get 512K memory map using this tool regardless of the memory size setting used when compiling applications like esp-link,  Even the pre-compiled user1.2048.new.5.bin  from the official espressIf AT release will only result in 512K memory map.

I don't have a problem mapping to 1M or 4M if I use another tool.

igrr, 2016-03-14:
Ok, please upload a binary generated for flash size above 512k, i will try flashing it and see if i can reproduce this.

It's really odd that you get this behaviour, because the same tool is used in Arduino IDE and we haven't had issue reports from the users regarding flash mapping issues.

ovaltineo, 2016-03-14:
Ok, here is the link to esp-link compiled for 1MB [https://drive.google.com/file/d/0BwavIEueSQXzbWhMOFhUYk1QQmM/view?usp=sharing](url).

igrr, 2016-05-10:
The binary header contains 512k setting:
```
$ dd if=user1.bin bs=1 count=4 2>/dev/null | xxd
0000000: ea04 0000
```
High four bits of the fourth byte contain flash size: 0 = 512K, 1 = 256K, 2 = 1M, 3 = 2M, 4 = 4M. Your binary generation process did not set correct flash size header. This usually happens when the author of software assumes that flash size setting will be handled later by esptool.py.

closed 2016-06-17 by igrr