igrr/esptool-ck #34: Difference between esptool-ck and esptool.py?

rayshobby, 2016-04-17:
This is really a question instead of an issue. I am curious who esptool-ck is different from esptool.py (other than the obvious differences in the command line parameter names)? The reason I am asking is that when I flashed a firmware (compiled in Arduino) that uses SPIFFS, with esptool.py the device always fails to mount SPIFFS, while with esptool-ck it works fine. I am curious to find out why. Thanks.

nissimzur, 2016-04-17:
Ray,

You can call my Skype nissim.test we can share experiment work on that

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Tel          0544 500635
Tel          03 5716699
Skype      <callto:nissim.test> nissim.test
Follow     <http://lnkd.in/TTnsGz> Linkedin  http://lnkd.in/TTnsGz
Email      nissim@elinistech.com

                <http://www.elinistech.com/> www.elinistech.com

igrr, 2016-04-17:
The issue you are seeing might be related to the way esptool.py modifies binary image header.
Esptool-ck sets the flash chip size field of the binary image header when the .bin file is generated. Esptool.py, on the other hand, modifies this field when .bin file is loaded into flash. I think you can work around this by passing correct flash size parameter to esptool.py when you upload.

tonyrui, 2016-04-20:
OK, that makes a lot of sense. I will try it later. Thanks for the explanation.

closed 2016-04-20 by rayshobby