igrr/esptool-ck #35: ck reset method not working

luggi, 2016-04-17:
I have connected my esp07-module as shown here: http://reflowster.com/resources/images/blog/esp8266/usbprogram.png and the ck bootloader reset method isn't working. 
Altough with esptool.py it works flawlessly. As i'd like to use the arduino ide for flashing instead of doing it manually it would be cool if this issue could be fixed :)
Here are the differences in the code: 
https://github.com/igrr/esptool-ck/blob/b56cb4c1e56e1c5f192cab73c0204f6bbe710c38/espcomm/espcomm_boards.c#L68 vs https://github.com/themadinventor/esptool/blob/master/esptool.py#L165
Thanks for your help.

nissimzur, 2016-04-17:
Leave rts (rst) line low is error. Has to be 1 after dtr(0)

void board_ck_rb()

	
{

	
    serialport_set_rts(1);

	
    serialport_set_dtr(1);

	
    serialport_send_break();

	
    espcomm_delay_ms(5);

	
    serialport_set_rts(0);

	
    espcomm_delay_ms(250);          // wait for break to finish

	
    serialport_set_dtr(0);

 

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Tel          0544 500635
Tel          03 5716699
Skype      <callto:nissim.test> nissim.test
Follow     <http://lnkd.in/TTnsGz> Linkedin  http://lnkd.in/TTnsGz
Email      nissim@elinistech.com

                <http://www.elinistech.com/> www.elinistech.com

igrr, 2016-05-10:
Well `serialport_set_dtr(0);` is not "set DTR line low", it's "deassert DTR line". Normal (non-asserted) state of DTR line is high.

@luggi I'm not sure what we are doing differently, but I'm able to upload with the schematic you linked. The only difference is that I also do have a pullup resistors on GPIO0 and on RST. Which OS and which serial adapter are you using?