igrr/esptool-ck #37: Use Cesanta flasher stub to improve uploading speed

AdrianEddy, 2016-06-21:
Take a look at this: https://blog.cesanta.com/esp8266-superfast-flashing-flash-esp-in-5-seconds
It would be great to incorporate their flasher stub into esptool-ck

igrr, 2016-06-21:
Hey, thanks, I'm aware of Cesanta's flasher.
My plan regarding the stub was slightly different though. We have added compression to ESP32 bootloader protocol, based on initial work in feature/compression branch of this project. The idea now is to write a stub for ESP8266, based on ESP32 bootloader code, to handle decompression, MD5, and baud rate selection. Then we are going to have feature parity between the new ESP32 protocol and ESP8266 protocol, stub included.
The stub will likely be done as an official Espressif release.

AdrianEddy, 2016-06-21:
Awesome, thank you. Looking forward to it