igrr/esptool-ck #38: Make fails on windows 10

yusijs, 2016-07-10:
So I've tried with both nmake from vs studio express 2015 and WinGNU make, and neither works. Any pointers on how to build this project?

WinGNU make
> make
> fatal: Not a git repository (or any of the parent directories): .git
> fatal: Not a git repository (or any of the parent directories): .git
> cc -std=gnu99 -Os -Wall  -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -DWINDOWS -DVERSION=\"\"  -c -o argparse/argparse.o argparse/argparse.c
> process_begin: CreateProcess(NULL, cc -std=gnu99 -Os -Wall -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm -DWINDOWS -DVERSION=\"\" -c -o argparse/argparse.o argparse/argparse.c, ...) failed.
> make (e=2): The system cannot find the file specified.
> make: *** [argparse/argparse.o] Error 2

nmake
> Microsoft (R) Program Maintenance Utility Version 14.00.23506.0
> Copyright (C) Microsoft Corporation.  All rights reserved.
> 
> makefile(1) : fatal error U1036: syntax error : too many names to left of '='
> Stop.

igrr, 2016-07-10:
It builds with MinGW, please see build steps in appveyor.yml file.

yusijs, 2016-07-10:
Fab, thanks, I'll give it a whirl!

closed 2016-07-10 by yusijs

yusijs, 2016-07-11:
If anyone else happens across this:
MinGW can compile this fine, but add CC=gcc after make all. Otherwise it'll fail with "cc: unknown command"