igrr/esptool-ck #44: Virus at esptool-0.4.9-win32.zip?

burner-, 2016-10-15:
Suddenly F-Secure remove esptool.exe 
I download https://github.com/igrr/esptool-ck/releases/download/0.4.9/esptool-0.4.9-win32.zip and unzip it and it remove also that. 
I upload exptool.exe to virustotal and result looks quite alarming https://www.virustotal.com/en/file/3b6691658dc47298f784a89321866e5519498fdc015aea27f9ad237667e799ab/analysis/1476526377/  
8 / 46 virus scanners detect is as troijan. Is is compiled with infected machine?

igrr, 2016-10-16:
Releases are built on Appveyor, that's an automated build in a container. Then it is directly uploaded to GitHub. Honestly, I have no idea where to start looking for a potential source of infection...

burner-, 2016-10-16:
Most probably it is then false positive. I sent that file to F-Secure. Hopefully they will can give more information why it is detected as troijan.

burner-, 2016-10-16:
I got answer from F-Secure. They say that it is false positive and they will make update to db soon.

davidparreira, 2016-11-20:
Cisco and its Advanced Malware Protection (AMP) product is also detecting it as malware... Unfortunately, you need a contract with Cisco to report it as a false positive.

sluzynsk, 2017-03-10:
I work for Cisco and I'll try to report it into that team.

I did an in-depth analysis with Cisco ThreatGrid and there are two reasons it is triggering as malware:
1 - the COFF header timestamp is set incorrectly - the compiler or linker isn't setting it to a current date
2- there are callback functions in thread local storage, which is apparently a technique malware uses to execute code prior to the debugger can attach to the process.

The second item is the larger issue in terms of why it is seen as malware.

I don't know that you have any control over either of those issues via build options, but that's why it's triggering.