igrr/esptool-ck #50: RTS question

ouki-wang, 2017-03-30:
have you tested a standard USB CDC device except cp2102 and ch340?
I expect to receive an RTS packet(21 22 02 00 00 00 00 00) to reset esp8266.
but I can not receive it.
I add the source code

serialport_set_rts(1);
serialport_set_dtr(0);

in function serialport_close()
. 
packet(21 22 02 00 00 00 00 00) will be sended.

nissimzur, 2017-03-30:
I see no attached files I can review the problem.

ouki-wang, 2017-03-31:
@nissimzur   what attached files are you refering to? download cmd line for esp8266 or others?

nissimzur, 2017-03-31:
You say "I add the source code"

Please add a full source code and I will test on my side. 
It is not clear where you add 
"iserialport_set_rts(1);  serialport_set_dtr(0); "
 in function serialport_close()

ouki-wang, 2017-03-31:
void serialport_drain(void)
{
    if(sPort)
    {
		FlushFileBuffers(sPort);
    }
	
}

int serialport_close(void)
{
    if(!sPort)
		return 0;
	serialport_drain();
	serialport_flush();
	serialport_set_rts(1);///////// my code
        serialport_set_dtr(0);///////// my code
	CloseHandle(sPort);
	sPort = NULL;
	return 1;
}

void serialport_set_dtr(unsigned char val)
{
    if(sPort)
    {
		EscapeCommFunction(sPort,((val)?SETDTR:CLRDTR));
    }
}

ouki-wang, 2017-03-31:
@nissimzur in serialport/serialport.c

igrr, 2017-03-31:
Sorry for a dumb question @ouki-wang, but why do you expect to receive an RTS packet when serial port is being closed? We aren't resetting the ESP using external reset, instead we are telling the ROM bootloader of the ESP to start running the application.

ouki-wang, 2017-03-31:
@igrr  I want to use a pin to control GPIO0 when receiving RTS signal。
ch340 and cp2102 can work very well，but they are vendor protocol。
how to control GPIO0 using standard usb cdc？

ouki-wang, 2017-03-31:
@igrr sorry ，I am not familiar with the CDC protocol。
I do not know which USB CDC command is used to control GPIO0. I would like to pull the GPIO0 high after the completion of the download, to prevent the case of continuous power, press the reset button again into the bootloader.
so，I want to pull the GPIO0 high after receiving RTS signal。

igrr, 2017-03-31:
GPIO0/CH_PD get toggled when the port is opened, call chain looks like this:

espcomm_open > espcomm_sync > espcomm_enter_boot > espcomm_board_reset_into_bootloader > (one of board_XX_rb functions in espcomm_boards.c).

There you can see that depending on reset method, different manipulations with RTS/DTR are performed. When serialport_set_rts is called, that will use ioctl (with TIOCMSET, TIOCM_RTS) (on Linux/Mac) or EscapeCommFunction (on Windows). Then it's the operating system's duty to translate this into underlying driver calls which should then emit corresponding packet.
If your problems lay below ioctl layer, I don't have anything I can help you, as this code only uses OS APIs and doesn't know anything about the USB protocol.

ouki-wang, 2017-03-31:
@igrr  thanks for your help。i test my cdc and ch340 and guess it is usb cdc windows driver‘s defect。

nissimzur, 2017-03-31:
Do you expect the state   serialport_set_rts(1);///////// my code serialport_set_dtr(0);///////// my code

Will be same after you kill the USB serial port ? 

The driver will free that to accept new connection. meaning RTS and DTR will change. 

Please work with single step and verify the voltage level on RTS DTR before USB serial port closed and after. 

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Skype      <callto:nissim.test> nissim.test
Follow     <https://goo.gl/CXY35s> Linkedin  https://goo.gl/CXY35s
Email      nissim@elinistech.com

                <http://www.elinistech.com/> www.elinistech.com