igrr/esptool-ck #52: Espcomm_sync failure Wemos D1R2

igrr, 2017-05-10:
_From @johnwa27 on May 10, 2017 2:51_

I'm getting this well known failure.  All the responses seem to place blame on the users code or wiring. But it's happening to me on a Wemos board with nothing connected to it.    It happens randomly, sometimes my code uploads, other times it will take 3 or 4 tries.  Closing the serial monitor seems to help a bit.  

This is not an issue with my code, my code works when it uploads and it happens with several different working programs.
It's a hardware/ide issue.  It's made even worse since it has to recompile every upload attempt.

When is someone going to take this issue seriously?

_Copied from original issue: esp8266/Arduino#3224_

igrr, 2017-05-10:
_From @davisonja on May 10, 2017 2:58_

I think it's fair to say that it is taken seriously, but it has a myriad of
potential causes meaning that there isn't one simple fix for everyone.

For example, I've got a "Wemos D1R2" board (I'm sure it's a clone, but
still) which has a flaky USB socket. It needs to be at just the right angle
in order for the board to appear on my computer as a serial port - the
symptoms, if the USB cable isn't just right are the 'espcomm_sync' errors.
Here it's an easy fault - the USB connector on the board isn't attached
properly, requiring pressure from the connected USB cable to complete the
connection.
Of course, it's a localised hardware fault to that board, so it's not
something that anyone else can either fix, or diagnose for me.

If you've got the same kind of issue you might need to talk to your
supplier.

igrr, 2017-05-10:
Moved to this repo, since the issue is with the esptool. I am sure there are a fair few cases where esptool-ck is "not trying hard enough".
It would be great if someone with access to the said board and capability to get logic analyzer traces and UART logs via a separate adaptor could get a trace for the case when failure occurs.

nissimzur, 2017-05-10:
Some Chines flashes are bad. Using them on PCB may result that. 

The rejected flashes sold in very low  prices and can use for FAT voice storage, not as Firmware flash and boot devices. 

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Tel          0544 500635
Tel          03 5716699
Skype      <callto:nissim.test> nissim.test
Follow     <https://goo.gl/CXY35s> Linkedin  https://goo.gl/CXY35s
Email      nissim@elinistech.com

                <http://www.elinistech.com/> www.elinistech.com

igrr, 2017-05-10:
> Some Chines flashes are bad. Using them on PCB may result that.

I have seen this as well, but this should not produce `espcomm_sync failure`, more likely a timeout waiting for flash erase to complete. This issue is specifically about the initial handshake failure.