igrr/esptool-ck #56: Erase?

beicnet, 2017-07-17:
Using latest bin: esptool-0.4.11-win32.zip

But how I see, I can't erase my WeMos D1 mini

```
esptool.exe -cp COM7 -cb 115200 -cd nodemcu -ce
warning: espcomm_send_command: didn't receive command response
error: espcomm_erase_flash: FLASH_DOWNLOAD_BEGIN failed
error: Invalid argument or value after -ce (argument #8)
```

btw...but I can flash it without any issue.

nissimzur, 2017-07-18:
I believe you can’t flash it too, based on the log you send

 

 

________

Kind regards,
Dr. Nissim Zur
Wireless long range IoT Artificial Intelligence platforms
Wi-Fi, Bluetooth Smart, Sub GHz
Consultant & Lecturer, 
Research IoT Projects

Tel          +972 544 500635 Israel
Tel          +972 3 5716699
Skype      <callto:nissim.test> nissim.test
Email       <mailto:nissim@telnt.com> nissim@telnt.com

web          <http://www.telnt.com/> www.telnt.com

Zoom       <https://zoom.us/download> https://zoom.us/download  

Zoom ID  <https://zoom.us/j/3717699114> https://zoom.us/j/3717699114

beicnet, 2017-07-18:
@nissimzur Yes, I can, I flashed it a few times yesterday and today too, but can't erase, I thinks its related to timing.

nissimzur, 2017-07-18:
Just send a bin file that has 0xFFFF all over.

www.telnt.com

pieman64, 2017-08-05:
I also found out that I can't erase but "flashing" is working of sorts.
The flash is accepted but the ESP doesn't seem to reboot into running mode. Will do further tests.
What I did find I was able to flash a general sketch to a "dead" ESP01 and then return to the Arduino IDE and now it's running with an OTA hostname of Lazarus. I have a few dead ESP07's and at least one dead WeMos so I will try to resurrect them too when I have time.

beicnet, 2017-08-05:
I figured out, that it will erase, but also will freeze the command prompt window for a while, is it a bug?

pieman64, 2017-08-15:
Tried again today and erasing flash still doesn't work. Argument #8 error as per @beicnet 
Anyone have a real fix?

beicnet, 2017-08-15:
@pieman64 Unfortunately, there is no explanation given for that issue yet.

nissimzur, 2017-08-15:
The chip does not "know" if it flashes a bin file filled of 0xFFFF or a code. 
So, if you can flash 0XFFFF ok or other data, it must flash good apps bin file. 
If it does not run, you have a bug in your bin. 
Call my SKype nissmi.test I will help

beicnet, 2017-08-15:
@nissimzur We are talking here about "erasing" chip with command line prompt