igrr/esptool-ck #61: problem programming ESP12F with -bm dio option

nm99, 2017-09-19:
I have been using esptool (v0.4.9) to program ESP12F modules for some months now without any problems. I am running the tool on Win 10.
A typical command line is:
esptool.exe -vv -cd nodemcu -cb 230400 -cp COM8 -ca 0x00000 -cf WiFi_A1k.ino.bin

Recently I got some new ESP12F modules and discovered that they would program but would not then reboot correctly. Further investigation showed that the new modules have flash memory from a different vendor (GigaDevice  instead of  WinBond).

I downloaded the Expressif flash download tool and discovered that the modules worked fine when programed in SPI DIO mode but not when programmed in QIO mode ( the default mode in esptool). 

I then tried using esptool with the -bm switch to define DIO mode:
esptool.exe -vv -cd nodemcu -cb 230400 -cp COM8 -bm dio -ca 0x00000 -cf WiFi_A1k.ino.bin
but although the output from esptool showed that dio mode had been selected, the end result was the same as before, module programmed but not booting.
I also tried:
esptool.exe -bm dio -vv -cp COM8 -cd nodemcu -cb 230400 -ca 0x00000 -cf WiFi_A10k.ino.bin in case the order of switches had an effect, but it made no difference.

I then tried using the python version of esptool (esptool.py) with DIO mode set and that worked fine.

Is there a bug with the -bm switch in esptool.exe or have I simply misunderstood how to use it?

nissimzur, 2017-09-19:
When some software tool is working OK on PCB A and not on PCB B, in most of the time the problem is with PCB B and not in the tool. 

Probably PCB B used other USB chip or has no proper boot setting. 

If PCB A and PCB B is 100% the same vendor, the production of PCB B has a hardware problem. 

 

All the best
__________________

Kind regards,
Dr. Nissim Zur
Wireless long range IoT Artificial Intelligence platforms
Wi-Fi, Bluetooth Smart, Sub GHz
Consultant & Lecturer, 
Research IoT Projects

Tel          +972 544 500635 Israel
Tel          +972 3 5716699
Skype      <callto:nissim.test> nissim.test
Email       <mailto:nissim@telnt.com> nissim@telnt.com

web          <http://www.telnt.com/> www.telnt.com

Zoom       <https://zoom.us/download> https://zoom.us/download  

Zoom ID  <https://zoom.us/j/3717699114> https://zoom.us/j/3717699114

igrr, 2017-09-19:
@nm99 Unlike esptool.py (and Espressif Flash Download Tools), esptool-ck does not modify the binary image header when flashing. In fact, `-bm` option only affects binary image generation, and is a no-op during flashing. If you want the binary to be compatible with DIO flash, you need to choose different settings when generating the binary file (likely in Arduino IDE tools menu).

nm99, 2017-09-19:
@igrr Thank you for a rapid and helpful response!
 I think what misled me was the message "setting flash mode from qio to dio" that appears as the command executes as in the attached snip. I will either choose a different setting for the binary file as you suggest or use esptool.py.
That answers my question fully so I'll close this off just now.

[image]

closed 2017-09-19 by nm99