igrr/esptool-ck #74: FLASH_DOWNLOAD_DATA Error

onlinexman, 2018-08-19:
Hello, i am stuck on a problem while trying to flash an esp32 from another microcontroller. 

- I use a prebuild image from the esptool
- The sequence of commands i use is: 
1. Enter the bootloader    
2. SYNC
3. SPI_ATTACH
4. SPI_SET_PARAMS
5. FLASH_BEGIN
6. FLASH_DATA --  and here it fails.
Data: 0x01
Data: 0x05
Data: 0x00
Data: 0x00
espcomm_cmd: cmd 0x03
espcomm_cmd: direction 0x01
espcomm_cmd: response 0x20120707

Has anyone had this issue and how can i possibly fix it?

igrr, 2018-08-19:
Hi @onlinexman, this tool works with esp8266, not esp32.

onlinexman, 2019-03-06:
Thank you for the answer. I was able to fix it. I was not sending one of the commands properly.

closed 2019-03-06 by onlinexman